blue-oil/blueoil #749: Test without GPUs

hadusam, 2020-01-23:
In my understanding, current Blueoil codes do not depend on whether running on GPU or not except distributed training.
So, we can test all our codes without GPU (except distributed training part, currently not tested).
If we decide to use CPU for test, because we have our own Dockerfile for Blueoil, we need to run an e2e test on docker image(GPU) created by Dockerfile at least once, I think. Fortunately, we have only one Dockerfile for tensorflow-gpu, and we can set `CUDA_VISIBLE_DEVICES=-1` to skip using GPU without fatal errors.
It is a little bit strange solution(using `tensorflow-gpu` with `CUDA_VISIBLE_DEVICES=-1`), but the requirements of tests are covered, and we are free from the rich GPU calculation environment.
Without GPUs, we can easily to parallelize tests to get results faster, this benefit is more important, I think 😄

hadusam, 2020-01-23:
Already changed `blueoil-test` to run without GPUs at #605 #742 
I would like to change `lmnet-test` to run without GPUs at #738

iizukak, 2020-01-23:
I agree with @hadusam.
We can run all of our test without GPU.

closed 2020-01-24 by hadusam