blue-oil/blueoil #762: Move and rename dlk to blueoil/converter

iizukak, 2020-01-27:
We want to provide Blueoil as just one python package.
But, current Blueoil have `dlk` as independent python package.
Our main package directory is `blueoil`, So we need to move scripts in `dlk` to `blueoil`.

Here is the how move dlk.

- `dlk/python/*` -> `blueoil/converter/*`
- `dlk/utils` -> `blueoil/converter/utils`
- `dlk/tests` -> `tests/converter`
- `dlk/.gitignore` -> `.gitignore` merge to root `.gitignore`
- `dlk/.gitmodules` -> removed
- `dlk/README.md` -> removed. Most of the text are deprecated
- `dlk/setup.py` -> `setup.py` merge to root `setup.py`, include some requirements.
- `dlk/setup.cfg` -> removed
- `dlk/hw/intel/de10_nano` -> `hw/intel/de10_nano`
- `dlk/docs` -> `docs/converter`
- `dlk/examples` -> `blueoil/converter/examples`

We also need to change python import path.
This is big issue. We need to do in several PRs.

iizukak, 2020-02-14:
@primenumber 
Is it ok to move dlk to blueoil directory?
I think it will conflict your project.

primenumber, 2020-02-14:
@iizukak OK!

iizukak, 2020-02-17:
We create `tests/unit` directory for unit tests.
This issue's structure is also changed.

iizukak, 2020-02-21:
@kchygoe Thanks. We can close this issue.

kchygoe, 2020-02-26:
👍

closed 2020-02-26 by kchygoe