blue-oil/blueoil #772: Pillow version has different resizing results

odoku, 2020-01-29:
There is a resize process in pre_process, but the default interpolation method differs depending on the Pillow version, so it seems to affect learning.

https://github.com/blue-oil/blueoil/blob/master/lmnet/lmnet/pre_processor.py#L38

Pillow v7.0.0
https://github.com/python-pillow/Pillow/blob/7.0.0/src/PIL/Image.py#L1786

Pillow v6.2.2
https://github.com/python-pillow/Pillow/blob/6.2.2/src/PIL/Image.py#L1830

I think it is good to set the default interpolation method of blueoil's resize function to BICUBIC.

```python
RESAMPLE_NEAREST = PIL.Image.NEAREST
RESAMPLE_BICUBIC = PIL.Image.BICUBIC

def resize(image, size=[256, 256], resumple=RESAMPLE_BICUBIC):
    ...

class Resize(Processor):
    def __init__(self, size, resample=RESAMPLE_BICUBIC):
        self.size = size
        self.resample = resample

    def __call__(self, image, mask=None, **kwargs):
        image = resize(image, size=self.size, resample=self.resample)
        if mask is not None:
            mask = resize(mask, size=self.size, resample=self.resample)
        return dict({'image': image, 'mask': mask}, **kwargs)
```

odoku, 2020-01-29:
It may be unnecessary because the version of Pillow is specified in requirement.txt.

https://github.com/blue-oil/blueoil/blob/master/requirements.txt#L6

joelN123, 2020-05-12:
Yes, that's a good point, I agree about this, it would be good to specify the type of resampling. Otherwise, in the future, we might upgrade Pillow and not realize that the default resampling type has changed.
But, so far, we have been using `resample=NEAREST` default, I guess? So maybe we should specify that one.
It's interesting, in [release note 2.7.0](https://pillow.readthedocs.io/en/stable/releasenotes/2.7.0.html#resize-performance), they say the quality of `NEAREST` is close to `BICUBIC`. But in [release note 7.0.0](https://pillow.readthedocs.io/en/stable/releasenotes/7.0.0.html#default-resampling-filter), they say `NEAREST` is fast but low quality.