blue-oil/blueoil #778: Verify Keypoint Detection Tutorial Documentation

iizukak, 2020-02-03:
https://docs.blue-oil.org/tutorial/index.html
https://docs.blue-oil.org/tutorial/image_keypoint_detection.html
https://docs.blue-oil.org/tutorial/run_fpga.html

oatawa1, 2020-02-05:
In the step 1.3.2 (https://docs.blue-oil.org/install/install.html#connect-to-the-fpga-board-on-your-pc), FPGA board do not response to the PC(keyboard) via serial connection under this command.
> sudo cu -l  /dev/[your_fpga_board]  -s 115200

The current environment
PC: MacBook Pro (16-inch, 2019) macOS Catalina version 10.15.3
FPGA: DE10-Nano (SoC FPGA) board

An alternative way to solve is to connect via screen instead.
>  screen /dev/[your_fpga_board] 115200

iizukak, 2020-02-05:
@oatawa1 Thanks. Please write this problem as indipendent issue.

oatawa1, 2020-02-05:
@iizukak Thanks. I create a new issue here.
[FPGA board do not response to the PC(keyboard) via serial connection #790](https://github.com/blue-oil/blueoil/issues/790)

oatawa1, 2020-02-06:
After I follows the tutorial in 2.4.2 and use abouve example, the configuration file name is not be keypoint_detection_demo.yml. I think it should be my_config.yml. 

Thus, I created the pull request here [#797 ](https://github.com/blue-oil/blueoil/pull/797)

oatawa1, 2020-02-13:
PR #827 solved this issue.

iizukak, 2020-02-13:
@oatawa1 Thanks. We can close this issue.

closed 2020-02-13 by iizukak