blue-oil/blueoil #804: The time measurement results are different between run.py and lm_*.elf

hadusam, 2020-02-07:
After #803, we can reduce the difference, but there are still gaps between `run.py` and `lm_*elf`.
We need to discuss and investigate this issue. 🤔 

## Test case
* Network: `LmnetV1Quantize`

## Run with FPGA on DE10-Nano
* run.py's inference: 16ms
* elf results: 3+9 = 12ms
```
root@blueoil-de10nano-04:~/masu_test/argo/export/save.ckpt-156250/32x32/output/python# python run.py --trial 1 -i ../../inference_test_data/raw_image.png -c ../models/meta.yaml -m ../models/lib/lib_fpga.so
...
INFO:__main__:Benchmark avg result(sec) for 1 trials: pre_process: 0.007635  inference: 0.016171 post_process: 0.016254  Total: 0.04006
root@blueoil-de10nano-04:~/masu_test/argo/export/save.ckpt-156250/32x32/output/python# head output/output.json {
    "benchmark": {
        "inference": 0.016170999999999935, 
        "post": 0.01625399999999999, 
        "pre": 0.007635000000000058, 
        "total": 0.040059999999999985
    }, 
    "classes": [
        {
            "id": 0, 
root@blueoil-de10nano-04:~/masu_test/argo/export/save.ckpt-156250/32x32/output/python# ../models/lib/lm_fpga.elf ../../inference_test_data/000_images_placeholder\:0.npy ../../inference_test_data/133_output\:0.npy 
-------------------------------------------------------------
Comparison: Default network test  succeeded!!!
-------------------------------------------------------------
TotalInitTime 3068,  sum:3.068ms
TotalRunTime 9425,  sum:9.425ms
...
```

## Run without FPGA on DE10-Nano
* run.py's inference: 36ms
* elf results: 19+1 = 20ms
```
root@blueoil-de10nano-04:~/masu_test/argo/export/save.ckpt-156250/32x32/output/python# python run.py --trial 1 -i ../../inference_test_data/raw_image.png -c ../models/meta.yaml -m ../models/lib/lib_arm.so
...
INFO:__main__:Benchmark avg result(sec) for 1 trials: pre_process: 0.007937  inference: 0.036363 post_process: 0.017636  Total: 0.061936
root@blueoil-de10nano-04:~/masu_test/argo/export/save.ckpt-156250/32x32/output/python# head output/output.json 
{
    "benchmark": {
        "inference": 0.036363000000000034, 
        "post": 0.017635999999999985, 
        "pre": 0.007936999999999972, 
        "total": 0.06193599999999999
    }, 
    "classes": [
        {
            "id": 0, 

root@blueoil-de10nano-04:~/masu_test/argo/export/save.ckpt-156250/32x32/output/python# ../models/lib/lm_arm.elf ../../inference_test_data/000_images_placeholder\:0.npy 
../../inference_test_data/133_output\:0.npy 
-------------------------------------------------------------
Comparison: Default network test  succeeded!!!
-------------------------------------------------------------
TotalInitTime 1026,  sum:1.026ms
TotalRunTime 19264,  sum:19.264ms
...
```

primenumber, 2020-02-07:
`time.clock`(Python2) or `time.process_time`(Python3) return CPU time, not elapsed time.
Inference routines are parallelized, so these functions are not suitable in this case.
We can use `time.time`(Python2) or `time.perf_counter`(Python3).

primenumber, 2020-02-07:
In `nn.run()`, `astype` copies ndarray and `np.zeros` allocate and initialize ndarray, these tasks affect inference time.

hadusam, 2020-02-10:
@primenumber 
Thank you soooo much for your marvelous advice!!!

I tried to use `time.time` and I got the correct results! 😄 

## Run with FPGA on DE10-Nano
Almost the same!
* run.py's inference: 9ms
* elf results: 9ms
```
root@blueoil-de10nano-04:~/masu_test/argo/export/save.ckpt-156250/32x32/output/python# head output/output.json
{
    "benchmark": {
        "inference": 0.00923299789428711, 
        "post": 0.016097068786621094, 
        "pre": 0.005796909332275391, 
        "total": 0.031126976013183594
    }, 
root@blueoil-de10nano-04:~/masu_test/argo/export/save.ckpt-156250/32x32/output/python# ../models/lib/lm_fpga.elf ../../inference_test_data/000_images_placeholder\:0.npy ../../inference_test_data/133_output\:0.npy 
-------------------------------------------------------------
Comparison: Default network test  succeeded!!!
-------------------------------------------------------------
TotalInitTime 3068,  sum:3.068ms
TotalRunTime 9425,  sum:9.425ms
...
```

## Run with FPGA on DE10-Nano
Almost the same!
* run.py's inference: 19ms
* elf results: 19ms

```
root@blueoil-de10nano-04:~/masu_test/argo/export/save.ckpt-156250/32x32/output/python# head output/output.json 
{
    "benchmark": {
        "inference": 0.01926279067993164, 
        "post": 0.03496980667114258, 
        "pre": 0.008594036102294922, 
        "total": 0.06282663345336914
    }, 
    "classes": [
        {
            "id": 0, 
root@blueoil-de10nano-04:~/masu_test/argo/export/save.ckpt-156250/32x32/output/python# ../models/lib/lm_arm.elf ../../inference_test_data/000_images_placeholder\:0.npy 
../../inference_test_data/133_output\:0.npy 
-------------------------------------------------------------
Comparison: Default network test  succeeded!!!
-------------------------------------------------------------
TotalInitTime 1026,  sum:1.026ms
TotalRunTime 19264,  sum:19.264ms
...
```

PR #816 (2020-02-12): Fix benchmark script to measure elapsed time correctly
Fix #804 (with this PR and #803 )
Please see the results in #804 .
closed 2020-02-13 by hadusam