blue-oil/blueoil #852: output_template is broken

iizukak, 2020-02-18:
I found some problem with `output_template`

These files are missing.
- [ ] missing `output_template/python/blueoil/data_augmentor.py`
- [ ] missing `output_template/python/blueoil/environment.py`
- [ ] missing `output_template/python/config.py`

tfujiwar, 2020-02-18:
I think `output_template/python/blueoil/data_augmentor.py` is not needed for inference.

iizukak, 2020-02-18:
@tfujiwar 
Answer is no...

```
$ python run.py -m ../models/lib/libdlk_x86.so -c ../models/meta.yaml -i /storage/iizuka/cifar/test/airplane/9439_airplane.png
INFO:__main__:Function _pre_process took 0.0009804419241845608 seconds to complete
INFO:__main__:Avg(func _pre_process): 0.0009804419241845608 sec.
INFO:__main__:Function _run took 0.049390214029699564 seconds to complete
INFO:__main__:Avg(func _run): 0.049390214029699564 sec.
INFO:__main__:Output: (before post process)
[[9.8979592e-01 2.6660317e-04 2.9553703e-04 3.0455622e-04 2.9522626e-04
  2.2118380e-05 5.4252555e-06 2.7977491e-03 1.7835555e-04 6.0384902e-03]]
Traceback (most recent call last):
  File "run.py", line 226, in <module>
    main()
  File "/home/iizuka/blueoil/venv/lib/python3.6/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/iizuka/blueoil/venv/lib/python3.6/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/iizuka/blueoil/venv/lib/python3.6/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/iizuka/blueoil/venv/lib/python3.6/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "run.py", line 216, in main
    run_prediction(input_image, model, config_file, trial=trial)
  File "run.py", line 147, in run_prediction
    output, bench_post = _timerfunc(_post_process, (output, config.POST_PROCESSOR), trial)
  File "run.py", line 112, in _timerfunc
    value = func(*extraArgs)
  File "run.py", line 46, in _post_process
    post_process = build_post_process(post_processor)
  File "/home/iizuka/blueoil/saved/cifar10_20200218112104/export/save.ckpt-78125/32x32/output/python/config.py", line 65, in build_post_process
    module = imp.load_module(module_name, f, pathname, description)
  File "/usr/lib/python3.6/imp.py", line 235, in load_module
    return load_source(name, filename, file)
  File "/usr/lib/python3.6/imp.py", line 172, in load_source
    module = _load(spec)
  File "<frozen importlib._bootstrap>", line 684, in _load
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/iizuka/blueoil/blueoil/post_processor.py", line 20, in <module>
    from blueoil.data_augmentor import iou
ModuleNotFoundError: No module named 'blueoil.data_augmentor'
```

tfujiwar, 2020-02-18:
@iizukak Sorry! Thanks for checking.

hadusam, 2020-02-19:
- [ ] missing `output_template/python/blueoil/post_processor.py`

And also we need an additional setting to support python2 on DE10-Nano.
- [ ] need to add `blueoil/utils/predict_output/__init__.py`
```
Traceback (most recent call last):
  File "run.py", line 27, in <module>
    from blueoil.utils.predict_output.output import JsonOutput, ImageFromJson
ImportError: No module named predict_output.output
```

- [ ] need to change `data_processor.py` support not only python3 but also python2 
```
Traceback (most recent call last):
  File "run.py", line 28, in <module>
    from config import (
  File "/root/LeapMind/TCAIPv2/face_detection/python/masu/python/config.py", line 23, in <module>
    from blueoil.data_processor import Sequence
  File "/root/LeapMind/TCAIPv2/face_detection/python/masu/python/blueoil/data_processor.py", line 68
    class Processor(metaclass=ABCMeta):
                             ^
SyntaxError: invalid syntax
```

iizukak, 2020-02-19:
@hadusam Wow, Thanks...

PR #871 (2020-02-28): Fix python files of output_template
## What this patch does to fix the issue.
Fix #852 

* Add `output_template/python/blueoil/post_processor.py` as a symbolic link
* Add `blueoil/utils/predict_output/__init__.py` to support python2 (DE10-Nano)
* Replace symlink of `data_processor.py` to the previous file (support python2) to revert a part of #832
  * https://github.com/blue-oil/blueoil/pull/871/commits/f88852d6f9fff54d8b5fc47e012f5f8d6401833c Just copied `data_processor.py` from [Before #832](https://github.com/blue-oil/blueoil/blob/bbe6c540d3ee4f80bed55c2c4150d4845c010d67/output_template/pyt [... cut]
closed 2020-02-28 by iizukak