blue-oil/blueoil #897: yaml parser error when train network

lm-lily, 2020-03-05:
[image]

I got yaml parser error when I tried to train network following Blueoil tutorial.
The error is as shown in the attached image.

iizukak, 2020-03-06:
@lm-lily THanks.
yaml file looks broken.
This problem looks like this post.
https://stackoverflow.com/questions/24983938/what-does-yaml-parser-parsererror-expected-document-start-but-found-blo

Can you put your yaml file?

IMO, It's good to copy and paste error message, not screenshot.

oatawa1, 2020-03-06:
@lm-lily Could you provide me more your `myconfig.yml`?

lm-lily, 2020-03-06:
@oatawa1 
I am sorry I don't know how to judge the format.
Could you tell me how to judge it?
I tried to attach the file contents here but it appears as weird formatting, so I attach the screenshot of the code.
[image]

oatawa1, 2020-03-06:
@lm-lily Thank you for the file. This is a python format. (I could see it from `import` part)

I'm not quite sure how to clearly judge it, in thia case I classified it by the header of the file.

oatawa1, 2020-03-06:
@lm-lily In case, you need to train it now, please manually change the configuration file extension to `.py`.

 Thank you for your point out. ^^ Now, I'm fixing the document.

lm-lily, 2020-03-06:
@oatawa1 
Thank you very much for the solution!

iizukak, 2020-03-09:
@lm-lily This issue can be closed?

closed 2020-03-09 by lm-lily