blue-oil/blueoil #898: Redundant lines are generated during user interaction in generating configuration file

lm-lily, 2020-03-05:
During user input for "[?] Please choose augmentors:", every click on any arrow button will generate a new duplicated line of "[?] Please choose augmentors:". 
[image]

iizukak, 2020-03-06:
@lm-lily Thanks. It's a bug.

iizukak, 2020-03-06:
@oatawa1 
It's looks related with #876 
Can you check?

oatawa1, 2020-03-06:
@lm-lily 
Thank you for reporting.
I think this is the problem of [`inquirer`](https://pypi.org/project/inquirer/) package. I hope they will fix it.

This happened when the line exceeds the window width.
Please extend the window width and try again.

closed 2020-04-15 by oatawa1