blue-oil/blueoil #909: lm_fpga.elf sometimes fails with classification

tk26eng, 2020-03-10:
An error occurs when we run lm_fpga.elf many times.
The probability that the problem occurs is around 1/1000.

In training, I almost followed the blueoil tutorial and modified some parameters.
The actual config is below.

```
# -*- coding: utf-8 -*-
# Copyright 2018 The Blueoil Authors. All Rights Reserved.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
# =============================================================================
from easydict import EasyDict
import tensorflow as tf

from blueoil.common import Tasks
from blueoil.networks.classification.lmnet_v1 import LmnetV1Quantize
from blueoil.datasets.image_folder import ImageFolderBase

from blueoil.data_processor import Sequence
from blueoil.pre_processor import (
    Resize,
    DivideBy255,
    PerImageStandardization
)
from blueoil.quantizations import (
    binary_mean_scaling_quantizer,
    linear_mid_tread_half_quantizer,
)

IS_DEBUG = False

NETWORK_CLASS = LmnetV1Quantize

# TODO(wakisaka): should be hidden. generate dataset class on the fly.
DATASET_CLASS = type('DATASET_CLASS', (ImageFolderBase,), {'extend_dir': '/home/blueoil/cifar/train/', 'validation_extend_dir': '/home/blueoil/cifar/test/'})

IMAGE_SIZE = [32, 32]
BATCH_SIZE = 64
DATA_FORMAT = "NHWC"
TASK = Tasks.CLASSIFICATION
CLASSES = ['airplane', 'automobile', 'bird', 'cat', 'deer', 'dog', 'frog', 'horse', 'ship', 'truck']

MAX_EPOCHS = 10
SAVE_CHECKPOINT_STEPS = 1000
KEEP_CHECKPOINT_MAX = 5
TEST_STEPS = 1000
SUMMARISE_STEPS = 100

# pretrain
IS_PRETRAIN = False
PRETRAIN_VARS = []
PRETRAIN_DIR = ""
PRETRAIN_FILE = ""

PRE_PROCESSOR = Sequence([
    Resize(size=IMAGE_SIZE),
    PerImageStandardization()
])
POST_PROCESSOR = None

NETWORK = EasyDict()

NETWORK.OPTIMIZER_CLASS = tf.train.MomentumOptimizer
NETWORK.OPTIMIZER_KWARGS = {'momentum': 0.9, 'learning_rate': 0.001}
NETWORK.LEARNING_RATE_FUNC = None
NETWORK.LEARNING_RATE_KWARGS = None

NETWORK.IMAGE_SIZE = IMAGE_SIZE
NETWORK.BATCH_SIZE = BATCH_SIZE
NETWORK.DATA_FORMAT = DATA_FORMAT
NETWORK.WEIGHT_DECAY_RATE = 0.0005

# quantize
NETWORK.ACTIVATION_QUANTIZER = linear_mid_tread_half_quantizer
NETWORK.ACTIVATION_QUANTIZER_KWARGS = {
    'bit': 2,
    'max_value': 2
}
NETWORK.WEIGHT_QUANTIZER = binary_mean_scaling_quantizer
NETWORK.WEIGHT_QUANTIZER_KWARGS = {}

# dataset
DATASET = EasyDict()
DATASET.BATCH_SIZE = BATCH_SIZE
DATASET.DATA_FORMAT = DATA_FORMAT
DATASET.PRE_PROCESSOR = PRE_PROCESSOR
DATASET.AUGMENTOR = Sequence([])
DATASET.ENABLE_PREFETCH = True
```

What is the problem ?

iizukak, 2020-03-10:
This is the critical bug...

tk26eng, 2020-03-13:
**Furtuer infomation after some investigation**
I tried to run `lm_fpga.elf` or `lm_arm.elf` under several condtions and recorded the probabilities.

* lm_fpga.elf (without any modification)
19/100000

* lm_arm.elf (without any modification)
0/100000

* lm_fpga.elf (OMP_NUM_THREADS=1)
167/100000

* lm_fpga.elf (using `/dev/mem` and OMP_NUM_THREADS=1)
0/100000

* lm_fpga.elf (using O_SYNC and OMP_NUM_THREADS=1)
0/100000

So this seems to be cache coherency problem.

iizukak, 2020-03-16:
@tk26eng 
Thanks you for debugging!!
It's good to find the way to avoid the bug.
Which is prefer `/dev/mem` or `O_SYNC` ?

primenumber, 2020-03-16:
It may be a bug of synchronization around CSR operations.
Can you check https://github.com/primenumber/blueoil/tree/insert-isb ?

iizukak, 2020-03-17:
@tk26eng Can you @primenumber suggestion? ?
https://github.com/blue-oil/blueoil/issues/909#issuecomment-599443783

tk26eng, 2020-03-19:
A result of `insert-isb` is here.
* lm_fpga.elf (applying `insert-isb` and OMP_NUM_THREADS=1)
28/100000

lm-watanabe, 2020-03-30:
@tk26eng Can you tell me the status of this issue?

tk26eng, 2020-03-30:
This problem came from bad cache operation and I made PR to fix this (just deleting one line).
This was software problem and not the hardware problem.

PR #957 (2020-03-31): Fix an issue which occurs invalid results with cache operation
## What this patch does to fix the issue.

## Link to any relevant issues or pull requests.
Fix #909
closed 2020-04-16 by tk26eng