blue-oil/blueoil #929: inference: run.py time measurement include library load

kchygoe, 2020-03-17:
v0.21.0

`output_template/python/run.py` can measure the time elapsed but it looks that first trial includes another process. 

```
...
[2020-03-17T16:32:43Z] INFO:__main__:Avg(func _pre_process): 0.0664009785652 sec.
--
  | [2020-03-17T16:32:49Z] INFO:__main__:Function _run took 6.28886604309 seconds to complete
  | [2020-03-17T16:32:49Z] INFO:__main__:Function _run took 0.611276865005 seconds to complete
  | [2020-03-17T16:32:50Z] INFO:__main__:Function _run took 0.614270925522 seconds to complete
...
```

iizukak, 2020-03-18:
@kchygoe Thank you for the report!!

tk26eng, 2020-04-20:
@kchygoe 
Now I 'm testing PRs with `.so`.
Your original problem is about when `.pb` file is read ?

The stability of measurements is getting improved with above PRs and it might be enough for `.so`.
But I'm suspecting `.pb` file still has problem.

kchygoe, 2020-04-22:
@tk26eng Thank you for working on this! You are right. Above data was by `.pb` file. (sorry, it should have been mentioned) I will check the details out

atsu-kan, 2020-05-15:
Is this issue still open?

tk26eng, 2020-05-15:
> Is this issue still open?

Oh, This is finished.
I close this.

PR #986 (2020-04-16): Remove some code from time measurement
## What this patch does to fix the issue
This code remove some code not to be included to time measurement results because it's part of initialization.
Even if this patch is applied, the first time measurement takes longer time than later results.
Please check the following issue to know more details of the problem.

## Link to any relevant issues or pull requests.
Fix #929
PR #1001 (2020-04-30): Fix a stability about multiple measurements
## What this patch does to fix the issue.
* This patch change the order to run the functions for multiple times because the continuous calling same function might estimate the consumed time lesser than actual situation
* This patch add the standard deviation value to know more statistics information about stabilities of measurement results

## Link to any relevant issues or pull requests.
Fix #929
closed 2020-05-15 by tk26eng