blue-oil/blueoil #933: Add dataset loader for FER2013

iizukak, 2020-03-18:
FER2013 is the dataset for Face Expression Recognition.
7 class classification.

It's looks better for tutorial than CIFAR10.

https://www.kaggle.com/c/challenges-in-representation-learning-facial-expression-recognition-challenge/data

kalpitthakkar-lm, 2020-05-22:
@iizukak I can work on this. Seems like a good issue to get warmed up.
Assigning myself to this.

kalpitthakkar-lm, 2020-05-22:
Ah, infact, FER2013 dataset images are not available anywhere. We need to find an alternative.

iizukak, 2020-05-25:
@kalpitthakkar-lm FER2013 dataset can be download from kaggle.
https://www.kaggle.com/c/challenges-in-representation-learning-facial-expression-recognition-challenge/data

kalpitthakkar-lm, 2020-05-25:
@iizukak Can you download it?
I am not able to access it even after login. It keeps showing an error in loading data.

iizukak, 2020-05-26:
@kalpitthakkar-lm 
Last month, I did download dataset. But now, there is not download link...

kalpitthakkar-lm, 2020-05-26:
@iizukak What should we do? Do you have any idea for alternative example?

How about this: http://vision.ucsd.edu/~iskwak/ExtYaleDatabase/ExtYaleB.html

iizukak, 2020-05-26:
@kalpitthakkar-lm 
Can you try to download?
https://www.kaggle.com/shawon10/facial-expression-detection-cnn/data

kalpitthakkar-lm, 2020-05-27:
@iizukak This does not have the images!

iizukak, 2020-05-27:
@kalpitthakkar-lm CSV contains image data

kalpitthakkar-lm, 2020-05-27:
@iizukak I got the dataset.

Okay, so how should I proceed?

iizukak, 2020-05-27:
@kalpitthakkar-lm Yes. Dataset loader.
metrics and loss, ... are already exist.

iizukak, 2020-08-24:
closed with #1172

closed 2020-08-24 by iizukak