blue-oil/blueoil #935: Tolerance in elf binaries should be variable as argument

tk26eng, 2020-03-18:
Now the tolerance to be tested in elf binaries like lm_x86.elf is fixed to `0.00001`. It should be variable because the tolerance is up to network and dataset.
Some developers try to choose the bigger tolerance in `blue-oil/converter/templates/include/dlk_test.h` and recompile it again when a calculated value and a ground truth doesn't match.
We can improve efficiency if we can change the tolerance as argument like below.

```
./lm_x86.elf [input.npy] [output.npy] [tolerance(optional)]
```

iizukak, 2020-03-19:
I want to know @primenumber 's opinion.

tk26eng, 2020-03-23:
I discussed with @primenumber about this directly.
This feature to change tolerance is acceptable for him.
But an warning should be shown when the user changes the tolerance because we might miss a bug with a bigger tolerance setting.

iizukak, 2020-03-24:
> But an warning should be shown when the user changes the tolerance because we might miss a bug with a bigger tolerance setting.

Sound good.