blue-oil/blueoil #937: The contents of keep_me.md should be placed at right place

tk26eng, 2020-03-19:
`blueoil/converter/templates/debug/keep_me.md` includes information about `-dbg` flag for `generate_project.py`. It should be placed at right place.

patrick-su, 2020-05-21:
@tk26eng do you mean we should be the content in `keep_me.md` to somewhere... like here?
https://github.com/blue-oil/blueoil/blob/master/docs/converter/

tk26eng, 2020-05-27:
@patrick-su 
Yes. That's right.
But maybe simply moving document is not enough because `debug` directory will be empty after moving `keep_me.md`.
Empty directory (`debug` directory) will be removed automatically by git command and no `debug` directory might occur problem for `generate_project.py`.
We need to consider also about this.