blue-oil/blueoil #940: Rule for C++ code of converter is little messy

tk26eng, 2020-03-23:
The files under `blueoil/converter/templates` has different styles.
It's better to be ruled to make reading the code easier.

Now following point I found is not ruled.

* Various style of include guard
There're several style in `.h` files.

1. pragma once style
```
# pragma once
```

2. traditional style
```
#ifndef `SOMETHING_UNIQUE`
#define `SOMETHING_UNIQUE`
#endif
```
A naming rule for SOMETHING_UNIQUE is also not ruled.

* Some `.h` includes definition
Some `.h` files include definition (implementation).
Definitions should be separated to `.cpp` file.
Otherwise the function defined in `.h` files can't be called from 2 or more places and a compilation error will happen.

* Deferent number of spaces between `.h` and `.cpp`
`.cpp` files have two spaces as a indent.
`.h` files have four spaces as a indent.
Code of `.cpp` have deep nest (indent) because of looping multi dimensional array, it might be acceptable.

What do you think about this ? Should it be improved ?

iizukak, 2020-03-24:
Yes, It's should be improved.
Should we decide the coding rule of C++?

primenumber, 2020-03-24:
pragma once style is not C++ standard, but accepted by gcc and clang.