blue-oil/blueoil #946: Can we remove a dataset subset type "train_validation_saving"?

tfujiwar, 2020-03-25:
A dataset subset type `train_validation_saving` makes `train.py` complicated. There are 6 if-statements with `use_train_validation_saving` in `train.py` and one of them has more than 40 lines.
https://github.com/blue-oil/blueoil/blob/969633c20aeea7ee964b2742efd5dfae331e6e80/blueoil/cmd/train.py#L256-L301

If we can remove this subset type, we can make `train.py` more readable (#481). Currently, this subset type is used only in cifar10 and cifar100.

joelN123, 2020-03-26:
I think it is a good idea to remove. This is specific to CIFAR mostly, so it is not ideal to make the main train code more complicated for it. Essentially, it allows saving the best model observed during training. This problem was somewhat specific to CIFAR because of the fluctuations of the model performance during training (more fluctuations compared to other datasets). However, since that time, other improvements such as [Final Sample Batch Normalization For Quantized Neural Networks](https://confit.atlas.jp/guide/event-img/jsai2019/2A4-E-2-05/public/pdf?type=in) give more stable training, so the problem is mostly mitigated.

tfujiwar, 2020-03-26:
@joelN123 Thank you for the comment! I've understood the purpose and the situation.

tfujiwar, 2020-03-27:
Removed.

closed 2020-03-27 by tfujiwar