blue-oil/blueoil #971: UserWarning for LmSegnetV1Quantize

iizukak, 2020-04-08:
I got UserWarning when convert `LmSegnetV1Quantize`.
Please update the network to fix this issue. 

```
Please see /home/blueoil/saved/segmentation_20200408012343/export/save.ckpt-917/360x480/output/README.md to run prediction
/home/blueoil/blueoil/converter/core/operators.py:60: UserWarning: WRN Output channels need to be multiple of 32 for space2depth1 of SpaceToDepth, but got output channel size of 12
  self._check_consistency()
```

tk26eng, 2020-04-09:
@iizukak 
I got same error messages as yours.
But the generated lm_x86.elf works well at my environment.
Is this just warning at your environment ?

iizukak, 2020-04-09:
@tk26eng 
Thanks.
Also work well with FPGA?

tk26eng, 2020-04-09:
Yes, the elf for FPGA also works well.
But anyway it's better to fix the warning.
I'll try to fix it.

iizukak, 2020-04-09:
I see. Thanks.

PR #982 (2020-04-16): Change condition to show an warning from generate_project.py
## What this patch does to fix the issue.
The number of output channel is checked if SpaceToDepth operator exists when `generate_project.py` runs.
The number of output channel for SpaceToDepth should be multiple of 32 in case of quantized situation.
But for floating point (not quantized) operation, the number of output channel doesn't have to be multiple of 32.

This PR removes the wrong checking of output channel for floating point operation.

## Link to any relevant issues or pull requests.
Fix #971
closed 2020-04-16 by tk26eng