blue-oil/blueoil #991: List up current network specification

iizukak, 2020-04-15:
We need the network specification list in the documentation

Target Network
- LmnetV1Quantize
- FYoloQuantize
- LmSegNetQuantize

Inference Target
- `lm_fpga.elf`
- `lm_arm.elf`

For each netork,
- Please train each network for just 10 epoch.
- Execute `profile_model.py`
- Convert the checkpoint
- And execute `lm_fpga.elf` and `lm_arm.elf`
- Put whole result of the documentation

## LmnetV1Quantize
- [ ] `profile_model.py`
- [ ] Inference Speed (FPGA)
- [ ] Inference Speed (Arm)

## FYoloQuantize
- [ ] `profile_model.py`
- [ ] Inference Speed (FPGA)
- [ ] Inference Speed (Arm)

## LmSegNetQuantize
- [ ] `profile_model.py`
- [ ] Inference Speed (FPGA)
- [ ] Inference Speed (Arm)

oatawa1, 2020-04-16:
## LmnetV1Quantize
- Inference Speed (FPGA)   **6.769** ms
- Inference Speed (Arm)    **16.526** ms
https://gist.github.com/oatawa1/ccd9f2d4879f7b2aace71a6448821242

## LMFYoloQuantize
- Inference Speed (FPGA) **59.011** ms
- Inference Speed (Arm) **141.915** ms
https://gist.github.com/oatawa1/2056a3b9242eff0853398b5c6bc8d51a

## LmSegnetV1Quantize
- Inference Speed (FPGA)      **394.196** ms
- Inference Speed (Arm)        **1431.74** ms
https://gist.github.com/oatawa1/90b899d43dfad2770baad42b12c309f7

## LmSegnetV1Quantize (Modify add preconvolution layer)
- Inference Speed (FPGA)        **400.509** ms
- Inference Speed (Arm)         **1437.04** ms
https://gist.github.com/oatawa1/78d261eded825315a9e46a086e961da8

iizukak, 2020-04-16:
@oatawa1 Thanks. Can you create document?

PR #1000 (2020-04-23): Add Network specification and update network code structure.
**This PR is ready. (#1004 merged)**

## What this patch does to fix the issue.
List network specification in the documentation. 
**Target Network**
- LmnetV1Quantize
- LMFYoloQuantize
- LmSegnetV1Quantize

**Inference Target**
- `lm_fpga.elf`
- `lm_arm.elf`

Also, update network code structure.

## Link to any relevant issues or pull requests.
Closes #991
closed 2020-04-23 by oatawa1