blue-oil/blueoil #1020: Verify C++ inference runtime

iizukak, 2020-05-04:
Currently, we have C++ inference code on `runtime` directory.
https://github.com/blue-oil/blueoil/tree/master/runtime

We are not sure that runtime can run correctly.

- [ ] Build for classification
- [ ] Check inference
- [ ] Build for object detection
- [ ] Check inference
- [ ] Update README.md

oatawa1, 2020-05-14:
I checked inference results in classification task, and I think the class label order maybe **invert**.
The input image is cat.
From class position of cat is in 4th.
But the inference result from softmax shows the highest score in 6th (which is maybe 10th-4th).

Results are here.
```
./run -i cat.npy -c model_classification/meta.yaml
classes:
airplane
automobile
bird
cat
deer
dog
frog
horse
ship
truck
task: IMAGE.CLASSIFICATION
expected input shape:
1
32
32
3
Run
Results !
shape:1 10
0.000527386 0.000344286 0.14302 0.0831289 0.0965507 0.00173161 0.673597 0.000440438 0.000252013 0.00040764
```

oatawa1, 2020-05-14:
PR #1030 update the README.md

oatawa1, 2020-05-14:
`PerImageStandardization` is not compatible with the current runtime.

I found that PerImageStandardization is already implemented in blueoil_data_processor.cpp but it’s not in mapping process in blueoil.cpp

Results in `PerImageStandardization is not register method:`

oatawa1, 2020-05-19:
Readme now updated in#1030 is merged. 
The problem about PerImageStandardization is solved in #1031 and merged.

So I close this issue.

PR #1030 (2020-05-18): Update (C++ inference) runtime documents
## What this patch does to fix the issue.
Update C++ inference runtime. e.g., change dlk -> blueoil converter.
Improve document understanding.

## Link to any relevant issues or pull requests.
Closes: #1020
closed 2020-05-19 by oatawa1