blue-oil/blueoil #1028: Cannot run inference again witout reboot after FPGA was stuck

hadusam, 2020-05-12:
When running different inferences at the same time, FPGA will be stuck and inferences are stopped. And after that, I cannot run any inference which needs FPGA, these will be stuck too. I know that the Blueoil does not support parallel inferences. I only would like to know how to reset the FPGA state without a reboot. Is there a way to reset this state?

### How to reproduce this situation
It is easy, run a following command on different terminal at the same time.
```
python run.py -i ../../inference_test_data/raw_image.png -c ../models/meta.yaml -m ../models/lib/libdlk_fpga.so --trial 100
```
It will be stopped on `INFO:__main__:Function _run` part.
And after that happened, we cannot run inference unless rebooting.

iizukak, 2020-05-12:
@hadusam Sorry, I don't know the solution.

@tk26eng Do you know something?

tk26eng, 2020-05-15:
Maybe there's no method to recover it after FPGA gets stucked.
But avoiding it is more important than recovering it.
We might need to implement the way to avoid multiple inference at same time at library level.
That is like singleton feature.