blue-oil/blueoil #1043: Random Seed for TF Record

minux302, 2020-05-21:
shuffle in `_TFDSReader` in [dataset_iterator.py]https://github.com/blue-oil/blueoil/blob/master/blueoil/datasets/dataset_iterator.py) has no random seed.
We must set random seed for shuffle for reproducibility

joelN123, 2020-05-22:
It's an important issue! Thank you for bringing it up.

Yes, no seed is passed, so it will be generated by the system clock maybe? Another reason there is variability between training runs is because the TensorFlow weight initializer's default is to use `None` random seed, which I guess has the same behaviour as `None` random seed for shuffle.

Speaking about reproducibility more generally, I think it is safer to generate random numbers by the system clock as much as possible (i.e. not use a fixed seed). For example, let's say I gave you a model definition and I say I got 70% test accuracy on it, for a particular dataset. Then you run training for it (with different seed) and the test accuracy is 65%. The cause of this big difference is something that we need to look into. One possible reason is that the training algorithm for the model has a large variability between runs. So it is important to try to reduce that variability. Or at least do multiple training runs, to get a mean and standard deviation. Another possible reason is that the test dataset is so small, the metric values fluctuate very easily. In either of these cases, we get a false sense of security by using the same fixed seed each time. In this sense, I think it is good to use the system clock for the random seed as default, so that these kinds of problems stay visible to us.

In the case where I'd like to give to someone my exact model, then I think it's better to just save the weights of the model, and let that person to load it / retrain it / etc as they like.

Having said all this, it would be useful to have an option for supplying a specific random seed for the training algorithm. For example, in debugging, if a bug comes up in a different place each time the training is run, or only comes up 10% of the time, it can be easier to debug by using the same fixed seed.

minux302, 2020-05-27:
Thank you for detail explanation!!
What is the implementation of creating a seed with a system clock  (not using a fixed seed)?
This is the implementation image. Is it correct?
```python
class _TFDSReader:

    def __init__(self, dataset, local_rank):
        tf.set_random_seed(0)
        tf_dataset = dataset.tf_dataset.shuffle(1024).repeat()
```

joelN123, 2020-05-28:
No problem!
- I think the TensorFlow default is to use the system clock as seed. So, simply not setting a seed (or setting as `None`) will cause the system clock to be used. I'm not totally sure how TensorFlow implements this... maybe somewhere inside of [tensorflow/core/lib/random](https://github.com/tensorflow/tensorflow/tree/r1.15/tensorflow/core/lib/random)?
- To use a fixed seed, yes, I think that adding the line `tf.set_random_seed(0)` will work (although I didn't try it out). That might also cause a fixed seed to be used for the weight initialization of the network [tf/random/set_seed](https://www.tensorflow.org/api_docs/python/tf/random/set_seed). To set a fixed seed for just the dataset shuffle (and not for other operations), it is possible to use `dataset.tf_dataset.shuffle(1024, seed=0).repeat()`.

After searching about this, it looks like even if the seed is set, there are some TensorFlow ops that may not give the same result when run on GPU [NVIDIA/tensorflow-determinism](https://github.com/NVIDIA/tensorflow-determinism) and also [Non-Determinism in TensorFlow ResNets](https://arxiv.org/abs/2001.11396)

minux302, 2020-05-29:
Thanks!
I want to, and will fix interface like below.
```python
class _TFDSReader:

    def __init__(self, dataset, local_rank, shuffle_seed=None):
        tf_dataset = dataset.tf_dataset.shuffle(1024, seed=shuffle_seed).repeat()
```

joelN123, 2020-05-29:
looks good ! nice and clean.