blue-oil/blueoil #1089: LmSegnetV1 structure

kalpitthakkar-lm, 2020-06-11:
I noticed that the current structure of the network in `lm_segnet_v1.py` does not follow the constraints of our IP (neither v2 or v3). Is this design on purpose?

To be specific, have a look at [this line of code](https://github.com/blue-oil/blueoil/blob/33edfc04c01ad19343946789fbb38d84ac45b3f1/blueoil/networks/segmentation/lm_segnet_v1.py#L66).
There is a `space_to_depth` call on the input?

Two problems:
1. Bad neural network design.
2. Input channels to `space_to_depth` not a multiple of 32.

The fixes would involve a conv layer (1x1) first that projects `channels=3` to `channels=32` and then putting a `space_to_depth`. Rest of the network is fine.

joelN123, 2020-06-12:
The input `images` is not quantized, so I guess that's why it doesn't cause any error. But, still, the weights of `conv1` are quantized. I'm not sure if there is a benefit of quantizing the weights of `conv1` when the inputs to it are not quantized.

In fact, in [lm_bisenet.py](https://github.com/blue-oil/blueoil/blob/33edfc04c01ad19343946789fbb38d84ac45b3f1/blueoil/networks/segmentation/lm_bisenet.py#L418), the weights of the first layer are not quantized. It could be good to do a similar thing as that model?

iizukak, 2020-06-12:
@kalpitthakkar-lm 
As Joel-san's comments. We do not quantize first layer.
We calculate first layer on CPU.

kalpitthakkar-lm, 2020-06-18:
> Umm, Now you are trying quantize all layers?

No I’m not quantising all layers. I just didn’t consider the fact that first layer is not quantized.

However, I still think that putting a space to depth as the first layer (before first convolution) is not a good idea.

kalpitthakkar-lm, 2020-06-22:
Closing this due to my doubts being solved, for now. Thank you @joelN123 @iizukak 😃

closed 2020-06-22 by kalpitthakkar-lm