blue-oil/blueoil #1102: Support dual inference at same time on FPGA

tk26eng, 2020-06-19:
Now runtime only support one inference on FPGA.
But sometimes running two models is useful.
We might need the feature for that.

primenumber, 2020-06-19:
Related issue: #666

kalpitthakkar-lm, 2020-06-22:
@tk26eng @primenumber 

Just a related question:
Is there a possibility of using multiple FPGAs for inference?
(Basically if the model is large, two FPGAs can be used for inference for two different models and the results are synced for combined output)

tk26eng, 2020-06-23:
@kalpitthakkar-lm 

> @tk26eng @primenumber
> 
> Just a related question:
> Is there a possibility of using multiple FPGAs for inference?
> (Basically if the model is large, two FPGAs can be used for inference for two different models and the results are synced for combined output)

Maybe we don't have any plan to use multiple FPGA for inference.
Bigger FPGA is easier way to handle large model with FPGA.