blue-oil/blueoil #1107: We should use relative comparing in testing model

tk26eng, 2020-06-21:
https://github.com/blue-oil/blueoil/blob/107c4fc71685159e6bfe077b263ac441d126b602/blueoil/converter/templates/include/dlk_test.h#L41-L57

In this above source, absolute error is used in comparing ground truth values and calculated values for elf binaries like `lm_x86.elf`. This comparison with absolute error works well when the values are normalized and range from 0.0 to 1.0.
But sometimes in some network, they are not normalized and range from 0 to 255 especially when output is image data. Then the absolute error of 0.00001 is too strict (small) for 32bit floating point which is valid within about 6 to 7 digit by decimal value.

Maybe, absolute error of 0.00001 is OK for lm_x86.elf because it doesn't use 32bit floating point.
But other elf binaries will use SIMD operations with 32bit floating point in some functions for faster calculation. So they might occur error of validation.

For example, the ground truth is 100.0 and calculated value is 100.000005, originally the test should pass. However, the comparison will fail because 32bit floating point has accuracy of 0.0001 (6digits of 100.0).

So we should use relative error for this problem.
This change from absolute error to relative one is very easy to modify. But this part has big effect to validation of model.

What do you think about changing absolute error to relative one ?

primenumber, 2020-06-22:
We should consider both relative and absolute error, because of loss of significance...

tk26eng, 2020-06-23:
@primenumber 
How about using threshold like this ?

```
float val_gt; // ground truth
float val_calc; // calculated value
// Certain value is set to val_gt and val_calc here
abs_tol = 0.00001; // Same threshold as original code for smaller value
rel_tol = 0.00001; // 6 digits should be matched for bigger value
if std::abs(val_calc - val_gt) < max(abs_tol, rel_tol * val_gt)
```