blue-oil/blueoil #1224: Stop to save config.yaml

iizukak, 2020-10-15:
Currently, we are saving `config.yaml` the contents is same as Python config.
IMO, nobody are using `config.yaml`.

https://github.com/blue-oil/blueoil/blob/master/blueoil/utils/config.py#L214-L236

oatawa1, 2020-10-15:
I think we may still need `config.yaml`.

To generate config.py, it duplicated the input python config file, while `config.yaml` is generated from the config dictionary.

It's possible that when we overwrite some variables, the `config.py` will not get update, but `config.yaml` will give us the correct configuration variables.

joelN123, 2020-10-15:
That's a good point, thank you for bringing it up.
It looks like there are two places where `config.yaml` file is saved
https://github.com/blue-oil/blueoil/blob/002f5408d11fd2a93457c5e7bc8dc876ace13ce0/blueoil/cmd/train.py#L299
and
https://github.com/blue-oil/blueoil/blob/002f5408d11fd2a93457c5e7bc8dc876ace13ce0/blueoil/cmd/export.py#L146
In `train.py`, it happens immediately after loading the config file. So I think there is no problem to not save the `config.yaml` there.
In `export.py`, it loads the config file, then overrides some things like `BATCH_SIZE` and `PRE_PROCESSOR`, before it then saves as `config.yaml`. But, after that, is the `config.yaml` used for anything? I understand that `meta.yaml` could be used for application such as demo. But I'm not sure about `config.yaml`.
The problem with not removing the `config.yaml` is that currently it can cause error when loading https://github.com/blue-oil/blueoil/pull/1110